pvasa/cameraview-ex #97: Video recording fails to save to file

yonathan06, 2019-01-05:
`stopVideoRecording()` always returns false, with no error

```
  private var tmpVideoFile: File? = null

  @SuppressLint("SimpleDateFormat")
  @Throws(IOException::class)
  private fun getNewVideoFile(): File {
    // Create an image file name
    val timeStamp: String = SimpleDateFormat("yyyyMMdd_HHmmss").format(Date())
    val directory = "${Environment.getExternalStorageDirectory().absolutePath}/filmit/videos".also { File(it).mkdirs() }
    return File(directory, "$timeStamp.mp4")
  }

  @SuppressLint("MissingPermission")
  private fun listenToVideoRecordEvent() {
    publisherDisposable = videoRecordEventPublisher.subscribe {
      if (it) {
        recordVideoFrame.visibility = View.VISIBLE
        tmpVideoFile = getNewVideoFile()
        recordVideoCamera.startVideoRecording(tmpVideoFile!!) {
          // Configure video (MediaRecorder) parameters
          audioEncoder = AudioEncoder.Aac
          videoFrameRate = 60
          videoStabilization = true
          videoEncoder = VideoEncoder.H263
        }
      } else {
        if(recordVideoCamera.stopVideoRecording()) {
          uploadTmpVideo()
        } else {
          Log.e(TAG, "failed to save video to file")
          Log.d(TAG, tmpVideoFile.toString())
        }
        recordVideoFrame.visibility = View.GONE
      }
    }

  }
```

pvasa, 2019-01-05:
Have you registered an error listener? All the errors are thrown to error listeners which can be registered as below 
```
CameraView.addCameraErrorListener { t: Throwable, errorLevel: ErrorLevel -> /* Camera error! */ }
```
Nothing can be implied from the information you provided above. Please reopen the issue when you have more information.

yonathan06, 2019-01-06:
I added the error listener, but it is never called (although the `stopVideoRecording` is still returning `false`)
any idea why?
any way to see all of CameraView logs?

yonathan06, 2019-01-06:
in logcat it gives me a lot of these: `E/mm-camera: <STATS_AEC ><ERROR> 1849: aec_process_calc_current_hist_metered_luma: error: out of range, exit`

yonathan06, 2019-01-07:
updated and `isVideoRecording` is `false` before calling `stopVideoRecording`
also saw that the interface of addCameraErrorListener has changed, but still no error
I'm setting `cameraMode` to `Modes.CameraMode.VIDEO_CAPTURE` programmatically

I'm running it on xiaomi mi 5 (android 8.0, API 26)

pvasa, 2019-01-09:
Lib using Timber does not change anything. Try logging or throwing the throwable that you get in camera error listener. Something like,
```
cameraView.addCameraErrorListener { t: Throwable, errorLevel: ErrorLevel ->
    Log.d(TAG, "Camera error, t) // If you are using Log api
    throw t
}
```

prajapatiravi257, 2019-01-09:
No error is thrown even after logging

pvasa, 2019-01-11:
@prajapatiravi257 are you using the same device as @yonathan06? If not what device are you on? If yes have you tested on other devices? This issue could be very specific to certain devices. Also try running the [sampleApp](https://github.com/pvasa/cameraview-ex/tree/master/sampleApp) on same device and see if you are still seeing same problem.

prajapatiravi257, 2019-01-11:
Yes even i have the same vendor model xiaomi 6A, I had also tested it on moto g5s plus and its working fine. And yes i am same issue with the sample app in xiaomi device.
Also even setting camera facing to front doesn't work.

keremkusmezer, 2019-01-12:
Can it be that the device is mapped as Camera1 because of the capabilities check?

prajapatiravi257, 2019-01-12:
yeah it can be but now i am using camera2 api from google samples for video recording and its working fine

pvasa, 2019-01-16:
It seems on certain devices, the library is falling back to camera1 implementation and thus could not start video recording (not supported on camera1). Implementing video recording on camera1 might be a future enhancement. In regards with this issue, the library functions as designed.

closed 2019-01-16 by pvasa