pvasa/cameraview-ex #99: Support for fixed orientation view and landscape video/image

pcm2a, 2019-01-06:
Currently in order to record a landscape video you have to leave the orientation unlocked. This results in a jarring experience when the device changes from portrait to landscape. A better approach would be to allow the device to be fixed to portrait or userPortrait. When the recording or picture is taken it would be taken in the correct orientation based on the device orientation.

Setting the activity to `android:screenOrientation="userPortrait"` locks the orientation but videos and images are taken in portrait even when rotated to landscape.

For Camera1 you would use camera.setDisplayOrientation, which is already being done in Camera1.kt.
For Camera2 I though it was supposed to just work by default since there is no setDisplayOrientation. However, it is not working.

I will start looking into how this is supposed to work and see if I can come up with a fix. If you have any tips they would be welcomed.

pcm2a, 2019-01-06:
Scratch my last attempt. Instead if you set `videoRequestBuilder.set(CaptureRequest.JPEG_ORIENTATION, 90)` around line 1296 in Camera2.kt, below the assignment of `videoRequestBuilder` then the preview will be correct and the video will be recorded in landscape.

What will need to happen is for both imageRequestBuilder and videoRequestBuilder the correct jpeg orientation will need to be calculated and set. If I get it working I will open a PR.

pcm2a, 2019-01-06:
Scratch everything above. It appears that the issue is with `outputOrientation`. The following line always returns the same thing if the device is locked to portrait `var sensorOrientation = cameraCharacteristics.get(CameraCharacteristics.SENSOR_ORIENTATION)`.

pcm2a, 2019-01-06:
I created PR https://github.com/pvasa/cameraview-ex/pull/100/files to fix this issue.

pvasa, 2019-01-06:
> I created PR https://github.com/pvasa/cameraview-ex/pull/100/files to fix this issue.

Closing this PR. https://github.com/pvasa/cameraview-ex/pull/100#issuecomment-451777388

pcm2a, 2019-01-13:
I didn't miss the merge of your refactor changes for this PR did I https://github.com/pvasa/cameraview-ex/pull/101?

> 6. Are you sure this is correct `((orientation + 45) / 90 * 90) % 360`?
In this equation everything is an integer. So orientation of 45 or less will end up as 0 % 360 = 0. Orientation of 46 will be 90 % 360 = 90, etc, etc. Seems to work really well in my testing.

pvasa, 2019-01-13:
> I didn't miss the merge of your refactor changes for this PR did I #101?

No I am still working on it. Also refactoring some background handling in the same PR which has some issues with mixing java threading and coroutines.

> In this equation everything is an integer. So orientation of 45 or less will end up as 0 % 360 = 0. Orientation of 46 will be 90 % 360 = 90, etc, etc. Seems to work really well in my testing.

You are right. Missed the fact that everything is an integer.

commit 52098c10 (2019-01-16): Fix fixed rotation problem. Fixes #99
PR #113 (2019-01-16): Bug fixes
- Better background task handling. Fixes #112 
- Fix rotation issue. Fixes #99
closed 2019-01-16 by pvasa