pvasa/cameraview-ex #104: AAPT linking references error

AngelStanislavovAngelov, 2019-01-08:
**Describe the bug**
Unable to build project when library is added. Getting the following error message

`/app/build/intermediates/incremental/mergeFreemiumDebugResources/merged.dir/values/values.xml:6167: error: expected boolean but got (raw string) continuous_picture.`

When I open the values.xml file the problem comes from the following line:
`<item name="autoFocus">continuous_picture</item>`

pvasa, 2019-01-09:
@AngelStanislavovAngelov What library version are you using? Did you recently update to latest lib versions? I think this is a cache issue. Try `./gradlew cleanBuildCache clean build --refresh-dependencies` on your Android project to clean, rebuild and redownload dependencies.

pedromarthon, 2019-09-24:
I am facing this issue