pvasa/cameraview-ex #106: Video recording and Frame processing simultaneously

bipinvaylu, 2019-01-10:
My use cases required to Preview, Record and Process the frames simultaneously, I have used this codebase, It seems promising but it's still, not supporting recording and frame processing at the same time. It would be nice if you can incorporate the same if you have sometime.

bipinvaylu, 2019-01-11:
Thanks @pvasa for a quick response, Yeah, Actually I am trying to achieve my use case since last few days. But unable to find any sustainable solution so far. Please find the proper use case details here: https://stackoverflow.com/questions/54091648/camera-2-api-to-preview-record-and-extract-the-preview-frame-to-firebase-face-d

Btw, If you can provide me some help/insight for the same. I can work out that way and if able achieve, I will push it here as well for others.

pvasa, 2019-01-27:
@bipinvaylu the way to achieve this would be to provide both `mediaRecorder` surface and `imageReader` surface to `videoRequestBuilder` and `camera?.createCaptureSession(..)`. So you will need to update `startVideoRecording()`.
You can take a shot at it and let me know if you need any help, meanwhile I am implementing (updating) the `CameraView.cameraMode` api to achieve exactly this functionality. With this update you will be able to enable multiple camera modes at the same time. For eg., `SINGLE_CAPTURE and CONTINUOUS_FRAME` or `VIDEO_CAPTURE and CONTINUOUS_FRAME` or even all 3 (possibly).

pvasa, 2019-01-29:
@bipinvaylu [Release v3.0.1-beta](https://github.com/pvasa/cameraview-ex/tree/v3.0.1-beta) now supports this feature.

closed 2019-01-29 by pvasa

krghv85, 2022-07-10:
Hi @pvasa 

This feature still is not working, it giving exception(java.lang.ArithmeticException: divide by zero) 'Unable to start video recording' 

Using - 3.5.5-alpha"

app:cameraMode="continuous_frame|video_capture"

I also log a issue for the same -  https://github.com/pvasa/cameraview-ex/issues/236