pvasa/cameraview-ex #110: Camera1 Video Recording Support Or Throw Not Supported Exception

keremkusmezer, 2019-01-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

keremkusmezer, 2019-01-12:
When the api falls back to Camera1 during VideoRecording, the startVideoRecording method is empty as well as the stop resume pause methods are returning false directly, do you have any plans to implement video recording for the Camera1 implementation and can't you just throw a NotSupported exception on startVideoRecording or execute the callback directly instead of leaving it empty?

pvasa, 2019-01-12:
There is no plan to include video support for camera1 any time soon. I will add an error for those methods.

closed 2019-01-12 by pvasa

pvasa, 2019-08-21:
Video capture is now supported.