pvasa/cameraview-ex #112: Camera.start causing crash intermittently

arvinkx, 2019-01-16:
**Describe the bug**
A clear and concise description of what the bug is.
Related to #96 - Same error, different scenario. I've tried using a version with the fix in that issue applied but am still running into this issue.

**To Reproduce**
Steps to reproduce the behavior:
-Starting the camera from `onRequestPermissionsResult` (even if using `runOnUIThread`) results in a crash due to `setValue` being called on a background thread. After allowing the permissions, the app crashes. The next launch of the camera (that doesn't happen in onRequestPermissionResult but in onResume) seems to be fine. 

**Expected behavior**
Camera should start without causing a crash.

**Using ProGuard**
- [x] with ProGuard
- [ ] without ProGuard

**Device (please complete the following information):**
 - Device: S8+
 - OS: Latest
 - API: 27
 - App version 0.1

**Additional context / stacktrace**
`019-01-15 16:25:32.573 24540-24638/com.speakezyapp.speakezyclient E/AndroidRuntime: FATAL EXCEPTION: CameraViewExBackground
    Process: com.speakezyapp.speakezyclient, PID: 24540
    java.lang.IllegalStateException: Cannot invoke setValue on a background thread
        at androidx.lifecycle.LiveData.assertMainThread(LiveData.java:443)
        at androidx.lifecycle.LiveData.setValue(LiveData.java:286)
        at androidx.lifecycle.MutableLiveData.setValue(MutableLiveData.java:33)
        at com.priyankvasa.android.cameraviewex.CameraConfigLiveData.setValue$cameraViewEx_release(CameraConfigLiveData.kt:33)
        at com.priyankvasa.android.cameraviewex.Camera2.updateOis(Camera2.kt:1010)
        at com.priyankvasa.android.cameraviewex.Camera2.access$updateOis(Camera2.kt:63)
        at com.priyankvasa.android.cameraviewex.Camera2$addObservers$$inlined$run$lambda$11.invoke(Camera2.kt:583)
        at com.priyankvasa.android.cameraviewex.Camera2$addObservers$$inlined$run$lambda$11.invoke(Camera2.kt:63)
        at com.priyankvasa.android.cameraviewex.CameraConfigLiveData$observe$1.onChanged(CameraConfigLiveData.kt:47)
        at androidx.lifecycle.LiveData.considerNotify(LiveData.java:113)
        at androidx.lifecycle.LiveData.dispatchingValue(LiveData.java:126)
        at androidx.lifecycle.LiveData$ObserverWrapper.activeStateChanged(LiveData.java:424)
        at androidx.lifecycle.LiveData$LifecycleBoundObserver.onStateChanged(LiveData.java:376)
        at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.java:355)
        at androidx.lifecycle.LifecycleRegistry.forwardPass(LifecycleRegistry.java:293)
        at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.java:333)
        at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.java:138)
        at androidx.lifecycle.LifecycleRegistry.markState(LifecycleRegistry.java:111)
        at com.priyankvasa.android.cameraviewex.Camera2.startPreviewCaptureSession(Camera2.kt:897)
        at com.priyankvasa.android.cameraviewex.Camera2.access$startPreviewCaptureSession(Camera2.kt:63)
        at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$1.onOpened(Camera2.kt:117)
        at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:139)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)`

**Camera Settings Applied**
`
<com.priyankvasa.android.cameraviewex.CameraView
                android:id="@+id/create_post_camera"
                android:layout_width="match_parent"
                android:layout_height="match_parent"
                android:adjustViewBounds="true"
                android:keepScreenOn="true"
                app:aspectRatio="4:3"
                app:autoFocus="continuous_picture"
                app:awb="auto"
                app:cameraMode="single_capture"
                app:facing="back"
                app:flash="auto"
                app:jpegQuality="high"
                app:noiseReduction="high_quality"
                app:opticalStabilization="true"
                app:outputFormat="jpeg"
                app:pinchToZoom="true"
                app:shutter="short_time"
                app:touchToFocus="true"
                app:zsl="true"
                app:layout_constraintLeft_toLeftOf="parent"
                app:layout_constraintRight_toRightOf="parent"
                app:layout_constraintTop_toTopOf="parent"
                app:layout_constraintBottom_toBottomOf="parent"
                />
`
Only other interaction with the library is to set the camera mode and `addPictureTakenListener` in `onCreate`. Let me know if any other information or data would help.

**Branch Name (if using as a local module)**
- [x] master
- [ ] development
- [ ] release
- [ ] other (please specify)

pvasa, 2019-01-16:
This will be fixed by #113

pvasa, 2019-01-18:
@arvinkx Ideally views should be updated on main (UI) thread. Technically this can be fixed in library and I will but even on app side, it should be kept in mind to deal with views only on main thread.

A quick fix would be to launch `camera.start()` on UI thread
```
override fun onRequestPermissionsResult(
        requestCode: Int,
        permissions: Array<out String>,
        grantResults: IntArray
) {
    // If permissions granted
    activity?.runOnUiThread { camera.start() }
}

```
You can use above fix until I update the lib.

closed 2019-01-18 by pvasa

arvinkx, 2019-01-18:
@pvasa I've already got the call to `camera.start` wrapped in `runOnUiThread` and I still am experiencing this issue. The sample app doesn't experience the issue and the only difference I can see is that the fragment with the camera is created after the permission result is returned and I am not using a fragment in the activity I have the camera in, I'm calling start from `onRequestPermissionsResult`. Our current workaround is to ask permission in a different activity.

PR #113 (2019-01-16): Bug fixes
- Better background task handling. Fixes #112 
- Fix rotation issue. Fixes #99