pvasa/cameraview-ex #116: How to capture images in 16:9

pcm2a, 2019-01-17:
Can you point me in the right direction on how to set the aspect ratio for capturing an image with Camera2? The equivalent of setPictureSize on Camera1. I would like to do something similar to what we did with videos and allow taking photos in 4:3 or 16:9. Currently with a full screen preview the captured image is 4:3.

pcm2a, 2019-01-17:
I figured this out and will update it to be similar to videos, allowing for Max 4:3 or Max 16:9.

`val largestPicture = pictureSizes.sizes(AspectRatio.Ratio16x9).last()`

pcm2a, 2019-01-17:
I finally figured it out, no code changes needed. The below setting, put in the camera opened listener seems to do the trick.
```
camera.addCameraOpenedListener {
   camera.aspectRatio = AspectRatio.Ratio16x9
 }
```

pvasa, 2019-01-18:
Not necessarily from opened listener. It can be set once the view is created using
```
camera.aspectRatio = AspectRatio.of("4:3") // any valid aspect ratio of format width:height
```
Also all supported aspect ratios can be found in `camera.supportedAspectRatios`

pcm2a, 2019-01-18:
I'll give that a shot also and report back. I tried setting app:aspectRatio="16:9" in the layout file but that seemed to give an error, maybe only works from the class file. Thanks for all the active development you are putting in to this project!

pcm2a, 2019-01-21:
Reopening to make sure I understand how 16:9 photos are supposed to be captured. The only way that seems to work good is in the opened listener.

pvasa, 2019-01-24:
With #134, one can specify the aspect ratio in xml directly like `app:aspectRatio="16:9"` and the camera preview will be initialized in that aspect ratio. Even the pictures will be taken in that aspect ratio.

PR #134 (2019-01-24): Fix aspectRatio is emitted to early. Update version.
Closes #116
closed 2019-01-24 by pvasa