pvasa/cameraview-ex #117: Capture fails after a restart of the camera

arvinkx, 2019-01-17:
**Describe the bug**
Capture doesn't work after stopping and restarting the camera in release 2.7.0. An error is not generated and the picture taken listener is not invoked. This showed up in this latest release (2.7.0), it doesn't seem to occur on 2.6.1.

**To Reproduce**
Steps to reproduce the behavior:
1. Start camera using `start`
2. Stop camera using `stop(false)`
3. Start camera again using `start`
4. Call `capture` and the dimming UI animation occurs but `addPictureTakenListener` or `addCameraErrorListener` are never called

**Expected behavior**
Calling capture should call either the error listener or the picture taken listener.

**Using ProGuard**
- [x] with ProGuard
- [ ] without ProGuard

**Device (please complete the following information):**
 - Device: S8+
 - OS: 8.0.0
 - API: 27
 - App version 0.1

**Branch Name (if using as a local module)**
- [ ] master
- [ ] development
- [x] release
- [ ] other (please specify)

pcm2a, 2019-01-17:
Is this reproducible in the sampleApp provided with the library?

arvinkx, 2019-01-17:
I can confirm this issue exists in the sample app as well by leaving the app and coming back. Flash will fire but the picture is not taken and no preview is shown.

pvasa, 2019-01-18:
> I had to include a google-services.json in order to compile it. I'm not sure why there is a need to have that in the project. @pvasa is that a dependency that could be removed? I also had to add gradle wrappers into a few sub projects.

@pcm2a @arvinkx The sampleApp demos barcode detection implementation using firebase api that is why it requires google-services.json. Please refer this [google help article](https://support.google.com/firebase/answer/7015592?hl=en) for instructions on generating google-services.json

arvinkx, 2019-01-18:
@pvasa Not really familiar with Kotlin but I believe the issue is related to this block of code in `CameraView.kt`:
```
override fun onPictureTaken(imageData: ByteArray) {
    coroutineScope.launch {
        pictureTakenListeners.forEach {
               it(imageData)
           }
      }
}
```
Once stop is called without removing the listeners, starting the camera again and calling capture in debug mode in the sample app, gets to the line with `pictureTakenListeners` but doesn't call `it(imageData)` when breakpoints are set. Hope this helps as I'm not really familiar enough with Kotlin to know why that block is not being called.

pvasa, 2019-01-20:
> I also do see this in the debugger: kotlinx.coroutines.JobCancellationException: Job was cancelled; job=JobImpl{Cancelled}@d922855

@arvinkx good catch that was the exact issue. When a scope is created using `CoroutineScope(coroutineContext)` function, it attaches the scope to a new `Job` context. So when `coroutineScope.cooutineContext.cancel()` is called it cancels the job and thus no more coroutines can be created using that scope.

PR #123 (2019-01-19): Fix coroutine scope
This patch fixes #117
PR #124 (2019-01-21): Coroutine scope fix with better api
- New Apis (fixes #117)
  - `CameraView.destroy()` - Tear down this `CameraView` instance. The instance is not usable anymore after this call.
  - `CameraView.isActive` - Check if this `CameraView` instance is still active and usable.
- More realistic sample app code
closed 2019-01-21 by pvasa