pvasa/cameraview-ex #119: Crash if launching CameraActivity multiple times

pcm2a, 2019-01-17:
**Describe the bug**
The latest batch of changes merged into developed has introduced an issue. If you launch your CameraActivity from another activity, then leave the camera, go back and forth a few times eventually it all locks up.

I have provided a simple activity that I used to reproduce the issue off of the development branch. When I switch to my fork that does not have your latest changes I am able to open the camera as many times as I want without issue.
**To Reproduce**
Pull request that demonstrates the issue:
https://github.com/pvasa/cameraview-ex/pull/118

1. Have an activity that you launch the camera from
2. Go back out of the camera
3. Go in and out of the camera a few times and it will fail

**Expected behavior**
Before the latest batch of changes the issue did not occur.

**Using ProGuard**
- [ ] with ProGuard
- [X ] without ProGuard

**Device (please complete the following information):**
 - Device: LG G6
 - OS: 8.0
 - App version [2.7.0]

pvasa, 2019-01-18:
@pcm2a I couldn't reproduce this behaviour using your demo branch #118 I also confirmed memory usage after closing and opening (also taking photos in between) camera activity multiple times and it seems normal. Let me know when you have more insights on this or more specific details about reproducing the issue.

pcm2a, 2019-01-18:
I will try to migrate the changes over in small batches and see if I can figure out when the issue appears. I will also monitor memory to see if that is what is causing the issue.

pcm2a, 2019-01-18:
I'm going to close this until I can provide any better information. I ran the profiler and memory and cpu both look great.

pcm2a, 2019-01-18:
I found the missing piece. If the layout has the activity starting in `video_capture` instead of `single_capture` mode then I get the crash. If set to `single_capture` I can open the camera unlimited times. I have updated the example.

Also if I try to launch in `single_capture` and then change to `video_capture` once the activity starts, eventually I run into the same issue. Maybe something is not getting closed/cleaned up when leaving the activity in video mode. 

I cannot reproduce this with a branch that is before the latest batch of merges.

pcm2a, 2019-01-19:
Another thing I have noticed. I can launch in `single_capture` mode, manually change to `video_capture` mode, exit, repeat, works every time.

pvasa, 2019-01-21:
@pcm2a I #123 and #124 might have fixed this issue. Can you please confirm?

pcm2a, 2019-01-21:
I am no longer able to reproduce the issue with the latest changes, thanks!

closed 2019-01-21 by pcm2a