pvasa/cameraview-ex #143: Not starting  on some devices

Wojtek1234, 2019-01-31:
Basic configuration of camera view does not start on some devices.

On android 9 One Plus 6 it works as it should (starting preview)

on Lenovo Yoga tab 3 Android 7.1 (api 25) and on samsung Samsung Sm-A500FU android 6.0.1 (api 23) CameraView falls into some strange behaviour and does nothing, 
in ovveriden method Camera2.start() it returns false and just hang there. In detail method chooseCameraIdByFacing() returns false so the true value from start is returned from line 626.

I have no idea why that is happening. Tested on both version 2.8.1 and 3.0.1-beta.

Basic Camera 2 api apps works on both devices.

CameraView widget has standart style from style="@style/Widget.CameraView"

pvasa, 2019-01-31:
Do you see any errors in camera view error listener?

Wojtek1234, 2019-02-01:
unfortunately no, app just hangs. It never comes to listeners, it hangs up on line 570 of CameraView class when it try to destroy camera object

pvasa, 2019-02-02:
@Wojtek1234 The reason `chooseCameraIdByFacing()` returns false could be because the hardware is not compatible with camera2 api on those devices. That is why it falls back to camera1 and return true which means the camera was started successfully.

Are you running sample app? If not, try the sample app and see if still facing same issue. Try adding a camera error listener first and then starting the camera. Please refer below snippet.
```kotlin
// add error listener first
cameraView.addCameraErrorListener { t, errorLevel -> 
   Log.e(javaClass.canonicalName, "Camera error", t)
}

// then start camera
cameraView.start()
```

ru-alxr, 2019-02-20:
Same bug.
Lib works perfectly on emulator, and fails on two real devices:
Lenovo a2016b30 (API 23)
Samsung SM-J810M (API 26)

Log contains these lines:
```
D/PermissionsManager: weaveJoinPoint: already has required permissions. Proceed with the execution.
I/CameraManagerGlobal: Connecting to camera service
D/VendorTagDescriptor: addVendorDescriptor: vendor tag id 3854507339 added
I/CameraManager: getCameraCharacteristics : cameraId = 0
I/CameraManager: getCameraCharacteristics : cameraId = 1
I/CameraManager: getCameraCharacteristics : cameraId = 0
```

pvasa, 2019-02-23:
@ru-alxr Thanks for posting the log, although it does not say anything about the actual issue. And I don't have hands on those devices right now so I cannot test them either.

There are some major improvements and features coming to camera1 implementation (which seems to be the problem here) of the lib. So I would suggest to look out for it and once the release is out, try running the new version on those devices. There is going to be a lot of bug fixes which might fix the issue that you and @Wojtek1234 are facing. Refer PR #147 for more info.

PR #149 (2019-02-26): Fix camera2 implementation which created issue in fallback mechanism
Fixes #143
closed 2019-02-26 by pvasa