pvasa/cameraview-ex #157: Provide access to CameraCharacteristics.SENSOR_ORIENTATION

petermolcany, 2019-03-08:
I'm trying to process frames in
`camera.setPreviewFrameListener { image: Image -> /* Preview frame available. */ }`
I managed to convert them to Bitmap however I'm having problems with rotation of the output images. I need to get the `CameraCharacteristics.SENSOR_ORIENTATION` irrelevant to device orientation to perform successful rotation.

As a workaround I do it now independently by calling `manager.getCameraCharacteristics(cameraId).get(CameraCharacteristics.SENSOR_ORIENTATION)`. Note `cameraId` is set manually as I could not find any way to get it from `CameraView`.

Please provide a clean way to get `CameraCharacteristics.SENSOR_ORIENTATION`.

pvasa, 2019-03-09:
@petermolcany I was thinking of replacing the `android.media.Image` with a custom data class very similar to `LegacyImage` but for camera2 impl. Can you help me understand your overall use case and how do you currently use the `Image` output? Would it be more useful to instead have `ByteArray` output for preview frames with metadata like width, height, and rotation (maybe more)? What are your thoughts?

petermolcany, 2019-03-11:
@pvasa I am using face recognition (draw overlay on preview) and simultaneous video recording. First, I need to convert preview frame to custom format to perform the detection. Library I am using requires the faces to be upright, sideways faces would not be detected. I need to know the camera sensor orientation to rotate the image and ensure faces are upright. Phone must be therefore held in portrait mode. (If person is lying on side I do not want the image to rotate due to device rotation)

Use case might be similar to one that is demonstrated in the sample. `FirebaseVisionImage.fromMediaImage(image, 0).detectBarcodes()` has hardcoded value of `0`. If detection function was orientation dependent it would require a proper way of passing rotation. Or if preview overlay was required.

I think the cleanest solution might be extension of the API with a getter for `cameraSensorOrientation`

In cases where detection should be device orientation aware `cameraSensorOrientation` in combination with `OrientationEventListener` could be used.

In most cases exposing `outputOrientation` would be sufficient but for my case where I just need the camera sensor orientation, this is not sufficient.

pvasa, 2019-03-15:
Orientation can now be accessed as below from version 3.2.0
```
// Inside picture taken listener and preview frame listener
outputImage.exifInterface.getAttribute(ExifInterface.TAG_ORIENTATION)
```
See `com.priyankvasa.android.cameraviewex.Image` for more information

PR #158 (2019-03-15): Consistent api across camera1 and camera2 implementations
- Make some API changes to make it consistent.
- Updated APIs (closes #157)
  - `addPictureTakenListener { image: Image -> }` 
      The output type has changed from ByteArray to Image which is a wrapper consisting of the original ByteArray image data and some extra metadata which has information about output orientation (more to be added later)
  - `setPreviewFrameListener { image: Image -> }`
     The `android.media.Image` is now replaced with internal Image data class same as the one in picture taken listener
  - `setLegacyPreviewFra [... cut]
closed 2019-03-15 by pvasa