pvasa/cameraview-ex #160: Crash on camera start

srivnaimish, 2019-03-19:
I have the camera in a fragment of a viewpager and have started and stopped the camera in onResume and onPause methods of the fragment. In some cases on start call getting a crash - 

Adding the stacktrace - 

Fatal Exception: java.util.NoSuchElementException: Sequence is empty.
       at kotlin.sequences.SequencesKt___SequencesKt.last(_Sequences.kt:202)
       at com.priyankvasa.android.cameraviewex.extension.SortedSetExtensionsKt$chooseOptimalPreviewSize$2.invokeSuspend(SortedSetExtensions.kt:44)
       at com.priyankvasa.android.cameraviewex.extension.SortedSetExtensionsKt$chooseOptimalPreviewSize$2.invoke(Unknown Source:10)
       at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:84)
       at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:142)
       at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
       at com.priyankvasa.android.cameraviewex.extension.SortedSetExtensionsKt.chooseOptimalPreviewSize(SortedSetExtensions.kt:33)
       at com.priyankvasa.android.cameraviewex.Camera2$startPreviewCaptureSession$2.invokeSuspend(Camera2.kt:858)
       at com.priyankvasa.android.cameraviewex.Camera2$startPreviewCaptureSession$2.invoke(Unknown Source:10)
       at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:84)
       at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:142)
       at kotlinx.coroutines.BuildersKt.withContext(Unknown Source:1)
       at com.priyankvasa.android.cameraviewex.Camera2.startPreviewCaptureSession(Camera2.kt:850)
       at com.priyankvasa.android.cameraviewex.Camera2$1$1.invokeSuspend(Camera2.kt:80)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:32)
       at kotlinx.coroutines.DispatchedTask.run(Dispatched.kt:236)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:594)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.access$runSafely(CoroutineScheduler.kt:60)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:742)

pvasa, 2019-03-19:
@srivnaimish would you mind adding some more info like lib version, device, etc. Try to fill the issue template as much as possible.

srivnaimish, 2019-03-19:
**Describe the bug**
App crash due to No element exception. Probably due absense of null check of sizes in camera2-  https://github.com/pvasa/cameraview-ex/blob/b54c3cc3a7635633cb427ed59f52b5778b22101f/cameraViewEx/src/main/api21/com/priyankvasa/android/cameraviewex/Camera2.kt#L904
Not sure though.

**Expected behavior**
Camera should start without causing a crash.

**Using ProGuard**
with ProGuard

Library version - 3.1.1

**Device :**

Device: Xiaomi POCO F1
OS: 9.0.0
API: 28

pvasa, 2019-03-19:
@srivnaimish please update the lib version to [3.2.0](https://github.com/pvasa/cameraview-ex/releases/tag/v3.2.0). Many such issues were fixed in this release. Reopen the issue if you still see those crashes after updating.

closed 2019-03-19 by pvasa

srivnaimish, 2019-03-22:
Hey @pvasa updated the library but still getting the crash.
**Stacktrace** - 
Fatal Exception: java.util.NoSuchElementException: Sequence is empty.
       at kotlin.sequences.SequencesKt___SequencesKt.last(_Sequences.kt:202)
       at com.priyankvasa.android.cameraviewex.extension.SortedSetExtensionsKt.chooseOptimalPreviewSize(SortedSetExtensions.kt:39)
       at com.priyankvasa.android.cameraviewex.Camera2.startPreview(Camera2.kt:904)
       at com.priyankvasa.android.cameraviewex.Camera2.access$startPreview(Camera2.kt:62)
       at com.priyankvasa.android.cameraviewex.Camera2$1.invoke(Camera2.kt:77)
       at com.priyankvasa.android.cameraviewex.Camera2$1.invoke(Camera2.kt:62)
       at com.priyankvasa.android.cameraviewex.TextureViewPreview$1.onSurfaceTextureSizeChanged(TextureViewPreview.kt:62)
       at android.view.TextureView.onSizeChanged(TextureView.java:372)
       at android.view.View.sizeChange(View.java:19984)
       at android.view.View.setFrame(View.java:19945)
       at android.view.View.layout(View.java:19848)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.RelativeLayout.onLayout(RelativeLayout.java:1083)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.support.v4.view.ViewPager.onLayout(ViewPager.java:1775)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.support.constraint.ConstraintLayout.onLayout(ConstraintLayout.java:1915)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.support.design.widget.HeaderScrollingViewBehavior.layoutChild(HeaderScrollingViewBehavior.java:142)
       at android.support.design.widget.ViewOffsetBehavior.onLayoutChild(ViewOffsetBehavior.java:41)
       at android.support.design.widget.AppBarLayout$ScrollingViewBehavior.onLayoutChild(AppBarLayout.java:1556)
       at android.support.design.widget.CoordinatorLayout.onLayout(CoordinatorLayout.java:888)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1816)
       at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1660)
       at android.widget.LinearLayout.onLayout(LinearLayout.java:1569)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.widget.FrameLayout.layoutChildren(FrameLayout.java:325)
       at android.widget.FrameLayout.onLayout(FrameLayout.java:261)
       at com.android.internal.policy.DecorView.onLayout(DecorView.java:894)
       at android.view.View.layout(View.java:19851)
       at android.view.ViewGroup.layout(ViewGroup.java:6141)
       at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:2729)
       at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2436)
       at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1558)
       at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7463)
       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1041)
       at android.view.Choreographer.doCallbacks(Choreographer.java:847)
       at android.view.Choreographer.doFrame(Choreographer.java:774)
       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1027)
       at android.os.Handler.handleCallback(Handler.java:809)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:166)
       at android.app.ActivityThread.main(ActivityThread.java:7555)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:469)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:963)

pvasa, 2019-03-23:
@srivnaimish Is it a possibility that you are facing this issue because of a similar cause described in #163. If so you can try doing a clean build and see if the issue is fixed. If not then please reopen this issue and we can investigate more.