pvasa/cameraview-ex #163: Camera 1 crash

srivnaimish, 2019-03-22:
**Describe the bug**
A crash on cameraview start on multiple number of devices in camera1. 

**To Reproduce**
Still in process of figuring out

**Expected behavior**
Camera should start normally

**Using ProGuard**
Yes

**Device (please complete the following information):**
 - Device: samsung ,xiaomi, lg, sony, others
 - OS: 4(98%), 6(2%)
 - API: 19,20,23
 - App version 2.1.6

**Additional context / stacktrace**
Fatal Exception: java.lang.VerifyError: com/priyankvasa/android/cameraviewex/VideoManager
       at com.priyankvasa.android.cameraviewex.Camera1$videoManager$2.invoke(Camera1.kt:69)
       at com.priyankvasa.android.cameraviewex.Camera1$videoManager$2.invoke(Camera1.kt:43)
       at kotlin.SynchronizedLazyImpl.getValue(LazyJVM.kt:74)
       at com.priyankvasa.android.cameraviewex.Camera1.getVideoManager(Camera1.kt)
       at com.priyankvasa.android.cameraviewex.Camera1.isVideoRecording(Camera1.kt:127)
       at com.priyankvasa.android.cameraviewex.CameraInterface$DefaultImpls.stop(CameraInterface.kt:45)
       at com.priyankvasa.android.cameraviewex.Camera1.stop(Camera1.kt:250)
       at com.priyankvasa.android.cameraviewex.CameraView.stop(CameraView.kt:564)
       at com.anurag.videous.fragments.defaults.camera.CameraFragment.onPause(CameraFragment.java:422)
       at android.support.v4.app.Fragment.performPause(Fragment.java:2645)
       at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1512)
       at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1784)
       at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1852)
       at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3269)
       at android.support.v4.app.FragmentManagerImpl.dispatchPause(FragmentManager.java:3245)
       at android.support.v4.app.Fragment.performPause(Fragment.java:2641)
       at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1512)
       at android.support.v4.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1784)
       at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1852)
       at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3269)
       at android.support.v4.app.FragmentManagerImpl.dispatchPause(FragmentManager.java:3245)
       at android.support.v4.app.FragmentController.dispatchPause(FragmentController.java:234)
       at android.support.v4.app.FragmentActivity.onPause(FragmentActivity.java:476)
       at com.core.activities.base.BaseActivityView.onPause(BaseActivityView.java:109)
       at com.core.activities.landing.LandingActivity.onPause(LandingActivity.java:433)
       at android.app.Activity.performPause(Activity.java:5546)
       at android.app.Instrumentation.callActivityOnPause(Instrumentation.java:1240)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3324)
       at android.app.ActivityThread.performPauseActivity(ActivityThread.java:3293)
       at android.app.ActivityThread.handlePauseActivity(ActivityThread.java:3271)
       at android.app.ActivityThread.access$1000(ActivityThread.java:166)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1294)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:136)
       at android.app.ActivityThread.main(ActivityThread.java:5584)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:515)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1268)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1084)
       at dalvik.system.NativeStart.main(NativeStart.java)

srivnaimish, 2019-03-22:
Crash happening on kitkat and marshmellow. 
In fragment
onResume starting the camera - cameraView.start();
onPause - cameraView.stop();
onDestroyView - cameraView.destroy();

Crash occurs when closing the app and relaunching it.

pvasa, 2019-03-23:
@srivnaimish Based on stacktrace it seems this exception is caused at class property initialization which suggests, it not being a functional error. On a second note, doing a quick google search I found [this SO post](https://stackoverflow.com/questions/100107/causes-of-getting-a-java-lang-verifyerror) which gives some insights of what the issue could be.
From the post, 
> `java.lang.VerifyError` can be the result when you have compiled against a different library than you are using at runtime.

I would suggest updating lib to latest version and doing a clean build using below command.
```
./gradlew cleanBuildCache clean assemble --refresh-dependencies
```

Also try running the sample app and see if you still face the same issue on those devices.

closed 2019-04-03 by pvasa