pvasa/cameraview-ex #174: Front facing camera doesn't fire addPictureTakenListener event

WilliamAlexander, 2019-04-10:
**Describe the bug**
Front facing camera doesn't fire addPictureTakenListener event

**To Reproduce**
Steps to reproduce the behavior:
1. Set camera to front facing
2. Call camera.capture()
3. The addPictureTakenListener does not fire
4. When camera is set to Back, it works fine and I'm able to save the image

**Expected behavior**
Save Image from addPictureTakenListener

**Using ProGuard**
- [ ] with ProGuard

**Screenshots**
n/a

**Device (please complete the following information):**
 - Device: [Google Pixel 2]
 - OS: [Android Oreo (9.0)]

**Branch Name (if using as a local module)**
- [ ] master

pvasa, 2019-04-11:
@WilliamAlexander Have you tried setting an error listener? If not please set `CameraView.addCameraErrorListener { throwable, level -> .. }` and see if it throws any error when you try to take picture from front camera. If there is some error it might give some insight about the issue.

WilliamAlexander, 2019-04-12:
@pvasa Thanks for the reply.
I have an error listener. 
I only get a warning:
"AF is not possible. Turning it off"

pvasa, 2019-04-15:
@WilliamAlexander the warning you're seeing only means auto focus is not supported and turned off. That shouldn't hinder other functionalities like taking pictures. Are you seeing same behavior while using sample app on the same device? Also please make sure that single capture mode is enabled; you can do that by checking `cameraView.isSingleCaptureModeEnabled`.

closed 2019-04-22 by pvasa