pvasa/cameraview-ex #182: Tap focus does not work

vladniki12, 2019-04-25:
**Describe the bug**
Tap focus does not work

**To Reproduce**
Steps to reproduce the behavior:
1. Add 'app:touchToFocus="true"' Inyo xml file
2. Tap on screen and focus not working

**Expected behavior**
Tap Focus

**Device (please complete the following information):**
 - Device: Redmi Mi A1
 - OS: Android ( 9.0)
 - API: API 28

**Branch Name (if using as a local module)**
- [ *] master
- [ ] development
- [ ] release
- [ ] other (please specify)

pvasa, 2019-04-25:
@vladniki12 does it not work at all or you still see the focusing circle but it does not focus correctly?

imdiegobf, 2019-05-24:
This is happening to me as well. I can see the focusing circle but does not focus properly.

rubensim, 2019-06-26:
Hi @pvasa, this is also happening on 
Pixel with Android 9
Samsung Galaxy S7 Edge with Android 8
Samsung galaxy S8, S6

Do you already know what is the issue ? and do you have any estimation on this ? this issue is very  critical for my App please respond as soon as you can.

pvasa, 2019-06-26:
Hi @rubensim i haven't had a chance to look into this issue. I'd suggest to use auto-focus for now if it is suffice for the use case.

askarsyzdykov, 2020-01-22:
Same issue 
Device: Xiaomi Mi A1
OS: Android ( 9.0)

askarsyzdykov, 2020-01-27:
FYI: Pinch to zoom does not work too