pvasa/cameraview-ex #185: Auto focus listener

xperiafan13-rom, 2019-04-29:
Hi, how i cad add listener for automatic focus without set points on startAutofocus?

pvasa, 2019-04-30:
Hi @xperiafan13-rom can you please elaborate the question? What kind of events are you trying to listen to?

xperiafan13-rom, 2019-05-01:
Soory, What I want is to implement a listener for an autofocus without needing some genre or using the startAutofocus function, that is, when doing automatic aoutocus to capture the image

pvasa, 2019-05-01:
@xperiafan13-rom I still don't get what you are looking but if you mean using automatic focus, you can set [`app:autoFocus="continuous_picture"`](https://github.com/pvasa/cameraview-ex#in-layout-xml) when defining camera view in your layout file. Or you can also set it programmatically by calling `cameraView.autoFocus = Modes.AutoFocus.AF_CONTINUOUS_PICTURE`.

closed 2019-05-03 by xperiafan13-rom