pvasa/cameraview-ex #186: How to get supported hardware level?

rid-hrant, 2019-05-02:
Hi. In Android camera 2 API there is implementation levels, I couldn't find any way, for checking implementation level. So how I can get implementation level?

#### Link to documentation https://developer.android.com/reference/android/hardware/camera2/CameraCharacteristics.html#INFO_SUPPORTED_HARDWARE_LEVEL

pvasa, 2019-05-02:
Hi @rid-hrant that information is not exposed. This library is meant to be a wrapper around camera implementation hiding all android apis behind a simple to use api. Internally the library checks for hardware level and if it is LEGACY it automatically fallsback to using Android camera1 api.

rid-hrant, 2019-05-02:
@pvasa thank you for fast feedback. How I can check is `YUV` output format supported or not?

pvasa, 2019-05-03:
@rid-hrant AFAIK YUV image output should be available on all devices. Although I haven't seen it mentioned anywhere in the docs but never have I either faced this issue where a device is not able to produce YUV images.

pvasa, 2019-05-03:
Also on devices with legacy hardware level, it will fallback to camera1 implementation which always produces `NV21` format images and it will be reflected in output `Image.format`.

rid-hrant, 2019-05-03:
In this [post](https://stackoverflow.com/a/48095179/7484338) there is mentioned about `Level 3` implementation, `Level 3: Additionally support YUV reprocessing and RAW image capture`. There can be device which does not have `Level 3` implementation, what about this case?

pvasa, 2019-05-04:
@rid-hrant YUV reprocessing is different than producing YUV output. Hardware level 3 check is required for [YUV reprocessing](https://developer.android.com/reference/android/hardware/camera2/CameraMetadata.html#REQUEST_AVAILABLE_CAPABILITIES_YUV_REPROCESSING) but YUV image output (YUV_420_888 or NV21) can simply be produced using [ImageReader](https://developer.android.com/reference/android/media/ImageReader) without any extra check.

rid-hrant, 2019-05-06:
@pvasa Is implementation of this method `camera.setPreviewFrameListener` related to [YUV reprocessing](https://developer.android.com/reference/android/hardware/camera2/CameraMetadata.html#REQUEST_AVAILABLE_CAPABILITIES_YUV_REPROCESSING)? If **YUV reprocessing** not supported, will it work?

pvasa, 2019-05-06:
No @rid-hrant it is not related to reprocessing.

rid-hrant, 2019-05-07:
@pvasa thank you! Now it's clear. 🙂

closed 2019-05-07 by rid-hrant