pvasa/cameraview-ex #187: It's not working on emulator

rid-hrant, 2019-05-07:
Hi. I have tried to take picture on emulator, it's not working!

pvasa, 2019-05-08:
@rid-hrant do you see any errors?

rid-hrant, 2019-05-08:
> @rid-hrant do you see any errors?

No.

rid-hrant, 2019-05-08:
Just black screen!
[image]

rid-hrant, 2019-05-08:
Code for taking picture.
`        mLayout.findViewById(R.id.capture).setOnClickListener(v -> mCameraView.capture());
`

pvasa, 2019-05-17:
@rid-hrant I found the cause of the issue. It is because Af mode 4 is not supported and when that happens the impl should fall back to `AF_OFF` but for some reason it does not and that is why always trying to use auto focus even when it is not supported. I haven't got a chance to look into the fix yet.

rid-hrant, 2019-05-17:
> @rid-hrant I found the cause of the issue. It is because Af mode 4 is not supported and when that happens the impl should fall back to `AF_OFF` but for some reason it does not and that is why always [...]

Good. Will wait for fix.