pvasa/cameraview-ex #191: About video recording

23nJay, 2019-05-28:
Hello, 
I'm beginner of android developer. I'm using this library because more interesting to integrate and easy. 

But how can i start to video recording like as capture Image ? 
Like it's easy coding to capture image 
cameraView.setCameraMode(Modes.CameraMode.SINGLE_CAPTURE);
cameraView.enableCameraMode(Modes.CameraMode.SINGLE_CAPTURE); 
cameraView.capture();

but there is no single syntax for video recording.. 
cameraView.setCameraMode(Modes.CameraMode.VIDEO_CAPTURE);
cameraView.enableCameraMode(Modes.CameraMode.VIDEO_CAPTURE);
cameraView.captureVideo();

Please! help me. 
Thanks.

pvasa, 2019-05-28:
Hi @23nJay you can use either `set..` or `enable..` which only prepares camera for that mode. In this case video mode. Then to actually start video recording, you can call `startVideoRecording(outputFile) { config.. }` which would start video recording. Please see [this section of doc](https://github.com/pvasa/cameraview-ex/blob/master/README.md#record-video) for more details.

23nJay, 2019-05-29:
But how can i declare "outputFile"? 
I declare like below.. 
File outputFile = null;
cameraView.startVideoRecording(outputFile);

and than it give an error >  "Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkParameterIsNotNull, parameter outputFile"

Could we start video recording direct without inserting any parameter? 

Thanks.

pvasa, 2019-05-29:
@23nJay No you cannot. outputFile is a required parameter. You must create a file like `File file = new File("path to the file")` and then pass it to start video recording. [Here are some options to access storage](https://developer.android.com/training/data-storage/files).

closed 2019-05-29 by pvasa

23nJay, 2019-06-06:
Hello, 
Finally i set like this:

cameraView.setCameraMode(Modes.CameraMode.VIDEO_CAPTURE);
cameraView.enableCameraMode(Modes.CameraMode.VIDEO_CAPTURE);
 
String root = Environment.getExternalStorageDirectory().toString();
File myDir = new File(root + "/cubeTok");
myDir.mkdirs();

String timeStamp = new SimpleDateFormat("yyyyMMdd_HHmmss").format(new Date());
String fname = "cubeTok_" + timeStamp + ".mp4";

File file = new File(myDir, fname);
cameraView.startVideoRecording(file);

But it's give an Exception: "com.priyankvasa.android.cameraviewex.CameraViewException: Unable to start video recording. 

Is there anything lake to put? Please me to solve this.

Thanks.