pvasa/cameraview-ex #192: Exception at Video recording

23nJay, 2019-06-07:
Hello,
Finally i set like this:

cameraView.setCameraMode(Modes.CameraMode.VIDEO_CAPTURE);
cameraView.enableCameraMode(Modes.CameraMode.VIDEO_CAPTURE);

String root = Environment.getExternalStorageDirectory().toString();
File myDir = new File(root + "/cubeTok");
myDir.mkdirs();

String timeStamp = new SimpleDateFormat("yyyyMMdd_HHmmss").format(new
Date());
String fname = "cubeTok_" + timeStamp + ".mp4";

File file = new File(myDir, fname);
cameraView.startVideoRecording(file);

But it's give an Exception:
"com.priyankvasa.android.cameraviewex.CameraViewException: Unable to start
video recording.

Is there anything lake to put? Please me to solve this.

Thanks.

pvasa, 2019-06-16:
@23nJay there's no more configuration to it. Can provide more details like full exception log and which device you're working with?

closed 2019-06-20 by pvasa

23nJay, 2019-06-22:
Hello, 

No there is no more details in exception, It comes when video starting method just called and Exception just shown in Toast method.

Kindly help, it's only one things thats I would like to implement. 

If  @pvasa wants my codes for understanding than i would to share, Please provide your secure email Id where can share my code files. 

Thank You @pvasa.

pvasa, 2019-06-22:
@23nJay can you share all camera related setup that you do in your layout file and code?

23nJay, 2019-06-24:
Hi @pvasa, 

Define in activity class: 
define global > CameraView cameraView;

and in onCreate() {
cameraView = (CameraView) findViewById(R.id.camera);

        if (cameraView != null) {
            cameraView.setFacing(FACING_BACK);
            cameraView.setAutoFocus(Modes.AutoFocus.AF_OFF);
            cameraView.setTouchToFocus(true);
            cameraView.setOpticalStabilization(false);
            cameraView.enableCameraMode(Modes.CameraMode.SINGLE_CAPTURE);
            cameraView.setCameraMode(Modes.CameraMode.SINGLE_CAPTURE);
            cameraView.start();
        }
}

And at .xml file 

 <com.priyankvasa.android.cameraviewex.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        android:keepScreenOn="true" />

Hope, it's may very help full. 
Thanks.

pvasa, 2019-06-24:
The `layout_width` or `layout_height` either of them should be `wrap_content`.

23nJay, 2019-08-12:
Hello,
Priyank Vasa,

We set layout_width = wrap_content and layout_height = match_parent but not
working for us, we would like to go with full screen of cameraView not in
wrap_content.
Where we set ayout_width = wrap_content and layout_height = match_parent
video recording work for us but it not support to pause, resume or stop in
this situation.

And Also we would like to know about that what is Camera( 01 or 02 )?