pvasa/cameraview-ex #194: [crash] Error opening camera with id 0

yangchendev, 2019-06-12:
**Describe the bug**
I have get much crash with Error opening camera with id 0

**Using ProGuard**
YES

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device (please complete the following information):**
 - Device: [Huawei Honor V20]
 - OS: [Android 9.0]

**Additional context / stacktrace**
Fatal Exception: com.priyankvasa.android.cameraviewex.CameraViewException: Error opening camera with id 0 (error: 4)
       at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$2$1.onError(Camera2.java:62)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:1922)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(CameraDeviceImpl.java)
       at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(-.java:8)
       at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:258)
       at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:182)
       at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:77)
       at android.os.Handler.handleCallback(Handler.java:907)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:216)
       at android.os.HandlerThread.run(HandlerThread.java:65)

yangchendev, 2019-06-15:
@pvasa

pvasa, 2019-06-16:
@yangchendev this seems to be like device specific issue. Have you tried running sample app on the same device?

rubensim, 2019-06-19:
Hi @pvasa , I got the same issue, on Pixel API 27 1080x1920:420dpi (Android 8.1 Google play) x86 emulator. 

com.priyankvasa.android.cameraviewex.CameraViewException: Error opening camera with id 0 (error: 4)
	at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$2$1.onError(Camera2.kt:149)
	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:1929)
	at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
	at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
	at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:258)
	at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:182)
	at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:77)
	at android.os.Handler.handleCallback(Handler.java:873)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:214)
	at android.os.HandlerThread.run(HandlerThread.java:65)
	
The error: 4 is the ERROR_CAMERA_DEVICE, please look [here](https://developer.android.com/reference/android/hardware/camera2/CameraDevice.StateCallback) for more details: 
----------------

I tried running Sample app as well, and got another exception:
com.priyankvasa.android.cameraviewex.debug E/CameraCaptureSession: Session 0: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2188)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:988)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:394)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl$2.onDisconnected(CameraCaptureSessionImpl.java:574)
        at android.hardware.camera2.impl.CameraDeviceImpl$7.run(CameraDeviceImpl.java:233)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
		
You can find more info [here](https://developer.android.com/reference/android/hardware/camera2/CameraAccessException
)

Could you please have a look and let me know if this is related to cameraview-ex and some fix should be done.

pvasa, 2019-06-19:
@rubensim the issues that you're seeing are because of using the lib on an emulator. The lib does not support emulator.

iulian0512, 2019-06-20:
i have this issue on  Galaxy S8 Android 9 

`Fatal Exception: com.priyankvasa.android.cameraviewex.CameraViewException: Error opening camera with id 0 (error: 3)
       at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$2$1.onError + 145(Camera2.kt:145)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError + 1929(CameraDeviceImpl.java:1929)
       at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source)
       at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept + 8(Unknown Source:8)
       at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke + 258(PooledLambdaImpl.java:258)
       at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke + 182(PooledLambdaImpl.java:182)
       at com.android.internal.util.function.pooled.OmniFunction.run + 77(OmniFunction.java:77)
       at android.os.Handler.handleCallback + 873(Handler.java:873)
       at android.os.Handler.dispatchMessage + 99(Handler.java:99)
       at android.os.Looper.loop + 214(Looper.java:214)
       at android.os.HandlerThread.run + 65(HandlerThread.java:65)`

layout file:

`<?xml version="1.0" encoding="utf-8"?>
<FrameLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:app="http://schemas.android.com/apk/res-auto"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    tools:context=".fragments.pv.pvocr.PvOcr">

    <com.priyankvasa.android.cameraviewex.CameraView

        android:id="@+id/fragpvocrcamera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:aspectRatio="4:3"
        app:autoFocus="continuous_picture"
        app:awb="auto"
        app:cameraMode="continuous_frame|single_capture"
        app:facing="back"
        app:flash="auto"
        app:jpegQuality="high"
        app:noiseReduction="high_quality"
        app:opticalStabilization="true"
        app:outputFormat="jpeg"
        app:pinchToZoom="false"
        app:shutter="short_time"
        app:touchToFocus="false"
        app:zsl="true" />

</FrameLayout>`

pvasa, 2019-06-20:
@iulian0512 in your layout file, either width or height should be `wrap_content`.

closed 2019-07-27 by stale[bot]

rocca23, 2019-09-02:
I've had some reports about this issue, devices are Samsung Galaxy A20e and Galaxy A40, both with Android 9. The OS version seems to be the common factor. In my case I had also errors with device id = 1, but always error code 4 (ERROR_CAMERA_DEVICE). The current workaround is to use `cameraView.addCameraErrorListener` to handle the error in a non-fatal way.