pvasa/cameraview-ex #195: IllegalStateException The specified child already has a parent. You must call removeView() on the child's parent first. com.priyankvasa.android.cameraviewex.PreviewImpl.markTouchAreas$cameraViewEx_release

iulian0512, 2019-06-16:
Fatal Exception: java.lang.IllegalStateException: The specified child already has a parent. You must call removeView() on the child's parent first.
       at android.view.ViewGroup.addViewInner(ViewGroup.java:4471)
       at android.view.ViewGroup.addView(ViewGroup.java:4312)
       at android.view.ViewGroup.addView(ViewGroup.java:4252)
       at android.view.ViewGroup.addView(ViewGroup.java:4225)
       at com.priyankvasa.android.cameraviewex.PreviewImpl.markTouchAreas$cameraViewEx_release(PreviewImpl.kt:85)
       at com.priyankvasa.android.cameraviewex.Camera2$previewSurfaceTappedListener$2$1.invoke(Camera2.kt:529)
       at com.priyankvasa.android.cameraviewex.Camera2$previewSurfaceTappedListener$2$1.invoke(Camera2.kt:70)
       at com.priyankvasa.android.cameraviewex.TextureViewPreview$$special$$inlined$apply$lambda$2.invoke(TextureViewPreview.kt:66)
       at com.priyankvasa.android.cameraviewex.TextureViewPreview$$special$$inlined$apply$lambda$2.invoke(TextureViewPreview.kt:34)
       at com.priyankvasa.android.cameraviewex.SurfaceTouchListener$tapGestureListener$1.onSingleTapConfirmed(SurfaceTouchListener.kt:46)
       at android.view.GestureDetector$GestureHandler.handleMessage(GestureDetector.java:303)
       at android.os.Handler.dispatchMessage(Handler.java:105)
       at android.os.Looper.loop(Looper.java:156)
       at android.app.ActivityThread.main(ActivityThread.java:6523)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:942)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:832)

pvasa, 2019-06-16:
@iulian0512 this is a known issue. It happens when you try to tap the preview too quickly. Current workaround is to block user from doing that. Denouncing the taps might help.

beom-mingyu, 2021-08-07:
me toooo