pvasa/cameraview-ex #196: 0 supported devices when deploying to Google Play Store

bsameh, 2019-07-01:
**Describe the bug**
Deploying an app that depends on this library to the Google Play Store renders the app unsupported on ALL Android devices. Removing the library as a dependency and any references to it in the code resolves the issue. According to the Google Play Store, 0 devices are supported due to the following issue:
[image]

This doesn't make sense since:
1. Devices that do support camera2 are still showing as unsupported with the same error.
2. We've included this in our AndroidManifest.xml file:
```Doesn't support required feature <uses-feature> - android.hardware.camera2```
This is to ensure that devices that do not support camera2 will still work.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload APK to Google Play Store
2. Check supported devices

**Expected behavior**
Devices should be supported, none are.

**Using ProGuard**
- [x] with ProGuard
- [ ] without ProGuard

**Screenshots**
Attached

**Device (please complete the following information):**
- All devices are impacted

**Additional context / stacktrace**
N/A

**Branch Name (if using as a local module)**
- [x] master
- [ ] development
- [ ] release
- [ ] other (please specify)

Any idea what we could be doing wrong? The error we receive from the Play Store seems to be misleading.

bsameh, 2019-07-01:
Here's the relevant snippet of my AndroidManifest.xml file:
```
<uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

    <uses-feature android:name="android.hardware.camera" />
    <uses-feature
        android:name="android.hardware.camera.autofocus"
        android:required="false" />
    <uses-feature
        android:name="android.hardware.camera2"
        android:required="false" />

    <uses-feature
        android:glEsVersion="0x00020000"
        android:required="false" />
```

pvasa, 2019-07-02:
@bsameh you don't need to put `uses-feature` for any camera feature. They are included in the library and will be merged to main manifest automatically when building the app.

pvasa, 2019-07-02:
Also please make sure you're using the latest version of the library.

closed 2019-07-07 by pvasa