pvasa/cameraview-ex #198: New logo/icon proposal

mansya, 2019-07-08:
Good day sir. I am a graphic designer and i am interested in designing a logo for your good project. I will be doing it as a gift for free. I just need your permission first before I begin my design. Hoping for your positive feedback. Thanks

pvasa, 2019-07-08:
Hi @mansya, good day. I am okay with it. What kind of logo do you have in mind?

mansya, 2019-07-11:
hi sir. I have completed the design for this project. How ? do you like it ? if yes which one do you like?
[image]

pvasa, 2019-07-12:
Hi @mansya thanks for putting them together. I liked the 3rd.

mansya, 2019-07-16:
hi sir. I have sent you [**Pull Requests**](https://github.com/pvasa/cameraview-ex/pull/200). please check. and I have also uploaded all files to this [**Drive**](https://drive.google.com/open?id=1tcwD5q3cKuzZn8qii6Hnesqut9blqkRU)

mansya, 2019-07-28:
Sir. How about the pull requests that I sent?

mansya, 2019-07-30:
I don't understand that. can you explain it?

closed 2019-09-05 by stale[bot]