pvasa/cameraview-ex #199: Library automatically adds RECORD_AUDIO

florianhirson, 2019-07-11:
**Describe the bug**
I use the library only to take pictures and the permission RECORD_AUDIO is automatically added to the manifest. 

Is there avoid this permission to be added ?

pvasa, 2019-07-12:
@florianhirson audio permission in Android requires explicit approval from user. So even if it is added in manifest, it won't be approved by default. You need to ask explicitly for runtime permission.

closed 2019-07-18 by pvasa

rocca23, 2019-08-30:
I think it's not the better approach to add all potentially used permission in the library manifest. Apps end up declaring permissions that they don't use and this may be relevant with Android < 6, where users accept permissions upfront. I think you should document what permissions are needed to use the relevant features and leave to developers the declaration of such permissions in their manifest.

For anyone looking into this issue, in the meanwhile you can remove unwanted permission (like the suspicious record audio permission) in this way:
```xml
<uses-permission
        android:name="android.permission.RECORD_AUDIO"
        tools:node="remove" />
```