pvasa/cameraview-ex #201: Stretched preview after camera switch within an activity with android:screenOrientation="portrait"

J3y-S, 2019-08-13:
**Describe the bug**
Hello. I'm using your very nice to use Camera API in an activity with android:screenOrientation="portrait".
When i start in portrait orientation and switch the camera after I've rotated the device to landscape orientation the preview ist totally stretched and wrong rotated.

**To Reproduce**
1. Start Activity with android:screenOrientation="portrait" including CameraView (hold the device in portrait orientation)
2. Rotate device to landscape
3. Switch Camera

**Expected behavior**
A non streched camera preview aber switching the camera

**Using ProGuard**
- [ ] with ProGuard
- [X] without ProGuard

**Screenshots**
*Device in portrait mode works very good!*
[image]

*Device rotated to landscape and camera switch twice (to front and again to back camera)*
[image]

I'm using the Version 3.4.3

pvasa, 2019-08-21:
Hi @J3y-S, I would suggest updating to `v3.5.4-alpha` and check if you still see this issue. A lot of preview and capture sizes related issues were fixed in some recent releases. If not then we can investigate more.

pvasa, 2019-08-21:
You can find complete changelog [here](https://github.com/pvasa/cameraview-ex/releases).

closed 2019-09-27 by stale[bot]