pvasa/cameraview-ex #212: Video and preview dark when recording with the front-facing camera

heyletscode, 2019-11-10:
**Describe the bug**
Video and preview turn dark when recording video with the front-facing camera.

Android Studio version: 3.5
API Version: 3.4.3

   `<com.priyankvasa.android.cameraviewex.CameraView
        android:id="@+id/cameraView"
        android:layout_width="150dp"
        android:layout_height="200dp"
        android:layout_margin="16dp"
        android:adjustViewBounds="true"
        android:keepScreenOn="true"
        app:aspectRatio="4:3"
        app:autoFocus="continuous_video"
        app:facing="front"
        app:opticalStabilization="true"
        android:layout_alignParentBottom="true"
        android:layout_alignParentEnd="true"/>`

This is all the configurations made and the recording starts normally as per the documentation

closed 2019-11-22 by heyletscode

JeetSA, 2020-01-20:
I am getting the same issue. Pleas help me out if anyone has solved it.

JeetSA, 2020-01-20:
@heyletscode I am getting the same issue. Can you please help me with that?

heyletscode, 2020-02-10:
@JeetSA I moved to Android's official camera library, that is CameraX

JeetSA, 2020-02-11:
@heyletscode  Okay thanks for reverting back.