pvasa/cameraview-ex #227: Method markState must be called on the main thread

EmreMetin, 2020-08-28:
When use switch , flash camera property, this error show.
FATAL EXCEPTION: CameraViewExBackground
    Process: com.example.photofilter, PID: 19964
    java.lang.IllegalStateException: Method markState must be called on the main thread
        at androidx.lifecycle.LifecycleRegistry.enforceMainThreadIfNeeded(LifecycleRegistry.java:317)
        at androidx.lifecycle.LifecycleRegistry.markState(LifecycleRegistry.java:109)
        at com.priyankvasa.android.cameraviewex.Camera2.startPreview(Camera2.kt:1131)
        at com.priyankvasa.android.cameraviewex.Camera2.access$startPreview(Camera2.kt:70)
        at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$2$1.onOpened(Camera2.kt:129)
        at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:139)
        at android.os.Handler.handleCallback(Handler.java:808)
        at android.os.Handler.dispatchMessage(Handler.java:101)
        at android.os.Looper.loop(Looper.java:166)
        at android.os.HandlerThread.run(HandlerThread.java:65)

DanielMartinus, 2020-09-11:
We're experiencing the same issue:

```
Fatal Exception: java.lang.IllegalStateException: Method markState must be called on the main thread
       at androidx.lifecycle.LifecycleRegistry.enforceMainThreadIfNeeded(LifecycleRegistry.java:317)
       at androidx.lifecycle.LifecycleRegistry.markState(LifecycleRegistry.java:109)
       at com.priyankvasa.android.cameraviewex.Camera2.startPreview(Camera2.kt:1131)
       at com.priyankvasa.android.cameraviewex.Camera2.access$startPreview(Camera2.kt:70)
       at com.priyankvasa.android.cameraviewex.Camera2$cameraDeviceCallback$2$1.onOpened(Camera2.kt:129)
       at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:150)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:237)
       at android.os.HandlerThread.run(HandlerThread.java:67)
```

HudsonGraeme, 2020-10-18:
I'm experiencing the same issue. @pvasa If you can offer any assistance here that would be much appreciated.

closed 2020-11-28 by stale[bot]